Nkawu/TFT_22_ILI9225 #7: Library won`t work on Arduino M0 board

SemVan, 2017-10-23:
Hi!
Could you please gimme the advice to change library code correctly to add Cortex-M0 support? 
I got Arduino M0 board, but according to Your source files and library.json it`s not supported.
Thanks.

Nkawu, 2017-10-23:
I updated the library to support all platforms, please let me know if it works

SemVan, 2017-10-24:
Actually it doesn`t
1.Warning message about "your board is not supported" dissappeared. So, this problem is solved, thanks a lot!
2. But 
SPDR,  SPSR, SPIF, _BV "were not declared in this scope"
I must admit that SPI.h, limits.h and other .h-files included in your lib are not presented in the same folder. Should they be there?
3. cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
     csport    = portOutputRegister(digitalPinToPort(_cs));
This cast seems to be really strange. 

Could You also please help me with this issues?

Nkawu, 2017-10-24:
It is impossible for me to help you with these issues without testing on the M0 board. Unfortunately I cannot afford to buy every model of Arduino board! I'll see if I can find any information.

SemVan, 2017-10-24:
There is no need in buying, it just doesn`t compile.
Anyway, thank You very much!

Nkawu, 2017-10-24:
I have spent some time looking at this, the code changes I made to make the TFT display a lot faster (bitbang) do not appear to be compatible with the Arduino M0. I don't have the knowledge or time right now to even attempt this. I have to remove the compatibility - sorry the library won't work with your board :(

Nkawu, 2017-10-26:
I have created a new version based on the Adafruit_ILI9341 library to support other Arduino boards. It is not and official release yet, but please try this and let me know if it works. You wll have to do a manual install into your libraries directory as shown in the Wiki:

[TFT_22_ILI9225-1.2.6.zip](https://github.com/Nkawu/TFT_22_ILI9225/files/1416856/TFT_22_ILI9225-1.2.6.zip)

SemVan, 2017-10-26:
Ypu`re a brick! 
It works fine, thnaks a lot!

Nkawu, 2017-10-26:
Good news, so the TFT display works 100%?

SemVan, 2017-10-26:
Absolutely.
Thanks once again, very useful job!

closed 2017-10-27 by Nkawu