Nkawu/TFT_22_ILI9225 #13: Triangle draw incorectly.

AndyILC, 2017-11-11:
Triangle draw incorectly.
try this sample:
  tft.drawTriangle(0,              tft.maxY()/2-1, tft.maxX()/2-1, tft.maxY()/2-1, tft.maxX()/2-1,0,              COLOR_WHITE);
  tft.drawTriangle(tft.maxX()/2-1, tft.maxY()/2-1, tft.maxX()-1,   tft.maxY()/2-1, tft.maxX()-1,0,                COLOR_BLUE);
  tft.drawTriangle(0,              tft.maxY()-1,   tft.maxX()/2-1, tft.maxY()-1,   tft.maxX()/2-1,tft.maxY()/2-1, COLOR_GREEN);
  tft.drawTriangle(tft.maxX()/2-1, tft.maxY()-1,   tft.maxX()-1,   tft.maxY()-1,   tft.maxX()-1,tft.maxY()/2-1,   COLOR_RED);

This problem occurs because of the use of the ABS() macro.
This fix fixes the problem.
//    int16_t steep = abs(y2 - y1) > abs(x2 - x1);
    int16_t y2y1 = y2 - y1;
    int16_t x2x1 = x2 - x1;
    int16_t steep = abs(y2y1) > abs(x2x1);
-------------
//    dy = abs(y2 - y1);
int16_t _y2y1 = y2 - y1;
    dy = abs(_y2y1);

Nkawu, 2017-11-12:
Thank you, I'll test it and make the change.

Nkawu, 2017-11-12:
I found that both the drawTriangle & fillTriangle functions had bugs. I fixed & tested both in v1.3.1. Thanks for reporting this!

closed 2017-11-12 by Nkawu