Nkawu/TFT_22_ILI9225 #16: using ILI9225 with STM32F1

nicolasimeoni, 2017-11-24:
Hi,
I tried to use the library with STM32F1, using the stm32duino framework, but to allow it to compile I had to modify some file, described below. Can you update your library so it can compile with stm32duino?
regards
Niko

file: TFT_22_ILI9225.h:
line 12: #if defined(ARDUINO_STM32_FEATHER) || defined(ARDUINO_ARCH_STM32) || defined(ARDUINO_ARCH_STM32F1) || defined(STM32F1)

file: DefaultFonts.c:
line 7: #if defined(ARDUINO_ARCH_SAM) || defined(ARDUINO_ARCH_SAMD) || defined(ARDUINO_ARCH_STM32) || defined(ARDUINO_ARCH_STM32F1) || defined(STM32F1)

file Basic_demo:
#if defined (ARDUINO_ARCH_STM32F1)
#define TFT_RST PA1
#define TFT_RS PA2
#define TFT_CS PA0 // SS
#define TFT_SDI PA7 // MOSI
#define TFT_CLK PA5 // SCK
#define TFT_LED 0 // 0 if wired to +5V directly
#else
#define TFT_RST 8
#define TFT_RS 9
#define TFT_CS 10 // SS
#define TFT_SDI 11 // MOSI
#define TFT_CLK 13 // SCK
#define TFT_LED 3 // 0 if wired to +5V directly
#endif

Nkawu, 2017-11-25:
Thank you Niko, I'll merge these changes into the code.

Nkawu, 2017-11-27:
I have added your changes to v1.3.3. Thanks!

closed 2017-11-27 by Nkawu