Nkawu/TFT_22_ILI9225 #17: ESP32 error

cypeks, 2017-11-25:
When I try to compile for the ESP32 example "Basic_Demo" I get an error:

`In file included from /home/cypek/Arduino/libraries/TFT_22_ILI9225/examples/Basic_Demo/Basic_Demo.ino:3:0:
/home/cypek/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.h:145:33: error: expected ';' at end of member declaration
         void begin(SPIClass &spi)
                                 ^
/home/cypek/Arduino/libraries/TFT_22_ILI9225/examples/Basic_Demo/Basic_Demo.ino: In function 'void setup()':
Basic_Demo:253: error: no matching function for call to 'TFT_22_ILI9225::begin()'
   tft.begin();
             ^
In file included from /home/cypek/Arduino/libraries/TFT_22_ILI9225/examples/Basic_Demo/Basic_Demo.ino:3:0:
/home/cypek/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.h:145:14: note: candidate: void TFT_22_ILI9225::begin(SPIClass&)
         void begin(SPIClass &spi)
              ^
/home/cypek/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.h:145:14: note:   candidate expects 1 argument, 0 provided
exit status 1
no matching function for call to 'TFT_22_ILI9225::begin()'`

My configuration:
Linux Mint 18.2 64bit
Arduino 1.8.5
board definition ESP32 espressif GIT version

Nkawu, 2017-11-25:
Thank you for reporting, will take a look at it

Nkawu, 2017-11-27:
I am unable to make the ESP32 board work in the Arduino IDE following the instructions at: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md. Unfortunately this means I cannot reproduce or fix the problem. I believe they will have support for the Arduino Board manager soon, at which point I will try to fix this.

cypeks, 2017-11-27:
I added the missing ";" in the TFT_22_ILI9225.h file on line 47 but there are further errors.

`/home/cypek/Arduino/Basic_Demo/Basic_Demo.ino: In function 'void setup()':
Basic_Demo:253: error: no matching function for call to 'TFT_22_ILI9225::begin()'
   tft.begin();
             ^
In file included from /home/cypek/Arduino/Basic_Demo/Basic_Demo.ino:3:0:
/home/cypek/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.h:145:14: note: candidate: void TFT_22_ILI9225::begin(SPIClass&)
         void begin(SPIClass &spi);
              ^
/home/cypek/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.h:145:14: note:   candidate expects 1 argument, 0 provided
exit status 1
no matching function for call to 'TFT_22_ILI9225::begin()'`

cypeks, 2017-11-27:
In the file "DefaultFonts.c" in line 5 after:
`#elif defined(ESP8266) || defined(ESP32)`
`#include <pgmspace.h>`
I added:
`#include <stdint.h>`
it eliminates the error: "unknown type name 'uint8_t'"

I did not really know what to do with the error:

/home/xxxx/Arduino/Basic_Demo/Basic_Demo.ino: In function 'void setup()':
Basic_Demo:253: error: no matching function for call to 'TFT_22_ILI9225::begin()'
   tft.begin();
             ^
In file included from /home/cypek/Arduino/Basic_Demo/Basic_Demo.ino:3:0:
/home/xxxx/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.h:145:14: note: candidate: void TFT_22_ILI9225::begin(SPIClass&)
         void begin(SPIClass &spi);
              ^
/home/xxxx/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.h:145:14: note:   candidate expects 1 argument, 0 provided
exit status 1
no matching function for call to 'TFT_22_ILI9225::begin()'

so just in the file "Basic_Demo" in line 253 I typed it:
`tft.begin(SPI);`

but I think I did something wrong because:

/home/xxxx/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.cpp:224:5: error: expected initializer before '_spi'
     _spi = spi;
     ^
/home/xxxx/Arduino/libraries/TFT_22_ILI9225/src/TFT_22_ILI9225.cpp:228:1: error: expected unqualified-id before '{' token
 {
 ^
exit status 1

Nkawu, 2017-11-27:
I managed to get stm32duino working on my laptop. The code compiles with v1.3.4, let me know if it works. Also please send me the pins you used for the TFT, I can add them to the example programs.

closed 2017-11-27 by Nkawu

Bolukan, 2018-09-27:
I solved one part (see below), but it is still not stable. Text leads to reboots. I managed to show Circle, fillRectangle and Line but just adding a second Line or Clear freezes the code. Maybe memory related?

```
SPIClass hspi = NULL;

// Setup
void setup() {
  Serial.begin(115200);
  Serial.println("Starting");
  hspi = SPIClass(HSPI);
  hspi.begin();
  tft.begin(hspi);
}
```