Nkawu/TFT_22_ILI9225 #21: compatible with 2.0" version

avandalen, 2018-01-16:
It seems that the 2.2 inch TFT is outdated and no longer being made. The 2.2" is hard to find now. More frequent you see the 2.0" TFT, which seems equal but smaller. Do you think that the library works for the 2.0"also?
https://www.ebay.com/itm/ILI9225-2-0-Inch-TFT-Display-Module-Serial-Port-SPI-Interface-Applied-176x220-/302188734615?hash=item465bda2897:g:r8sAAOSw5cNYb0Lf

Nkawu, 2018-01-16:
The resolution is the same at 176x220, so I don't see why it wouldn't work

avandalen, 2018-01-17:
Yes I see no problems eather

From: Johan Cronje 
Sent: Tuesday, January 16, 2018 8:07 PM
To: Nkawu/TFT_22_ILI9225 
Cc: Albert van Dalen ; Author 
Subject: Re: [Nkawu/TFT_22_ILI9225] compatible with 2.0" version (#21)

The resolution is the same at 176x220, so I don't see why it wouldn't work

closed 2018-02-28 by Nkawu