Nkawu/TFT_22_ILI9225 #22: Working config with esp8266

dariocdj, 2018-02-13:
Hello, i'm trying to make it work with an esp8266 from a couple of hours.
Please can you give me a working GPIO config between esp-12 and ili9225 ? thanks
Dario

Nkawu, 2018-02-13:
Hi Dario. Unfortunately I do not have access to an ESP8266 to test the display. I will order one and try to make it work.

dariocdj, 2018-02-13:
Argh, but you use with esp32 ? can you give me pin for esp32 ?

dariocdj, 2018-02-13:
Something is wrong... i've attached direct to an arduino uno and load demo, but doesn't work, white display

dariocdj, 2018-02-13:
Uhm pheraps fake display ? With softSPI it works.....  :(

dariocdj, 2018-02-13:
With Arduino everything work, also spi (not soft)... so display is working....

Nkawu, 2018-02-28:
I just released v1.3.5 of the library which has changes for ESP32 to make it faster. As I said before, I don't have a ESP32 to test, so I can't help you with regards to the SPI freq.

closed 2018-07-18 by Nkawu