Nkawu/TFT_22_ILI9225 #26: error when try to compile for esp8266

etscho1st, 2018-03-14:
Hi 
i have problem when i try to compile and uploat the test sketch i get following error message.

C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp: In member function 'void TFT_22_ILI9225::_setWindow(uint16_t, uint16_t, uint16_t, uint16_t, autoIncMode_t)':

C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:451:26: error: no matching function for call to 'min(uint16_t&, int)'

     x0 = min( x0,_maxX-1 );

                          ^

C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:451:26: note: candidates are:

In file included from c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.h:20,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:1:

c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)

     min(initializer_list<_Tp> __l, _Compare __comp)

     ^

c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note:   template argument deduction/substitution failed:

C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:451:26: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'

     x0 = min( x0,_maxX-1 );

                          ^

In file included from c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.h:20,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:1:

c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)

     min(initializer_list<_Tp> __l)

     ^

c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note:   template argument deduction/substitution failed:

C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:451:26: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'

     x0 = min( x0,_maxX-1 );

                          ^

In file included from c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.h:20,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:1:

c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^
[... 296 lines cut ...]

In file included from c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.h:20,

                 from C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:1:

c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)

     min(const _Tp& __a, const _Tp& __b)

     ^

c:\users\arduino\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:

C:\Users\Arduino\Documents\Arduino\libraries\TFT_22_ILI9225\src\TFT_22_ILI9225.cpp:454:26: note:   deduced conflicting types for parameter 'const _Tp' ('short unsigned int' and 'int')

     y1 = min( y1,_maxY-1 );

                          ^

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Nkawu, 2018-03-16:
I will take a look at this & let you know

Nkawu, 2018-03-18:
What version of the Arduino IDE and the ILI9225 library are you using? On my computer the Basic_Demo example compiles with no errors. I have IDE v1.8.3 and library v1.3.6 installed

etscho1st, 2018-03-18:
I have Arduino 1.8.5 and ili9225 1.36 lib.

Nkawu, 2018-03-18:
I upgraded my Arduino IDE to 1.8.5 but it still did not get the compilation error. When I updated my "esp8266 by ESP8266 Community" package in Boards Manager to v2.4.1, I get the same compilation error you get. The short term solution is to downgrade the version of your esp8266 package in Boards Manager until I get this issue resolved. Not sure if it's their problem or mine.

Nkawu, 2018-03-19:
It was a small change on my side to fix the compile error, so I just released v1.3.7 of the ILI9225 library. It may take a while to show up in the Arduino Library Manager.

closed 2018-03-19 by Nkawu

etscho1st, 2018-03-19:
K thx will test and report. 

⁣Gesendet mit BlueMail ​