Nkawu/TFT_22_ILI9225 #28: How to fade backlight?

antknight, 2018-07-01:
Thanks for creating this library! One thing I am struggling with is the backlight. I have tried various things to get the backlight to fade in. It seems all I can do is have it on or off. For example when analog write goes over 127 and backlight come on then when it goes below 128 it turns off. Do you think you could possibly provide an example sketch in the library demonstrating this?

PeratX, 2018-07-18:
Need that too

Nkawu, 2018-07-18:
Another contributor added this functionality to the library. I will try to test it as soon as I can to verify.

PeratX, 2018-07-19:
You should connect the LED pin to a hardware PWM pin.
Then just use the TFT_22_ILI9225::setBacklightBrightness() to change it