Nkawu/TFT_22_ILI9225 #35: Basic Example hangs on ESP8266

RanTalbott, 2018-10-14:
Testing on a Wemos D1 Mini, with the Opensmart 2.0" 176x220 LCD. 

Arduino version 1.8.4, ESP8266 board support V2.3.0, TFT library v1.3.8

The hardware works fine with the example from https://github.com/jorgegarciadev/TFT_22_ILI9225 but that library is older, and missing some features (most notably bitmap display), so I'd like to get it working with this one.

The example code starts up okay, but hangs and gets killed by the watchdog timer in the first execution of tft.clear().

I'm _guessing_ that there's some obscure bug/incompatibility with the 2.3.0 ESP8266 support. But that's just a guess. What version(s) has this library been tested with? Are there any that are required/should be avoided?

Thanks,

Ran

Bolukan, 2018-10-31:
You mean it crashes [here](https://github.com/Nkawu/TFT_22_ILI9225/blob/master/examples/Basic_Demo/Basic_Demo.ino#L274) after showing "hello!" on the screen?

I tried once with the ESP32, but find some time to connect a D1mini.

Bolukan, 2018-11-04:
I used the most recent version (30 sep 2018) of this library with an ESP32 and ILI9225, but without use of WiFi.  
It's on my list to test it with the ESP8266, of course I will report any remarks here.

See below for the relevant parts of code for the ESP32. I connected the blacklight to 5V: To use variable blacklight I guess I'll need some PWM code on the IO pin. I did not check whether the library supports it.
```C
// ILI9225 (new variant) -> ESP32
// 3V3 -> 3V3
// 5V -> not attached to ESP32
// GND -> GND
// SCK -> IO 14
// SDO -> IO 12
// SDI -> IO 13
// SS  -> IO 15
// RS -> IO 25
// RST -> IO 26
// BL -> 5V
```
```C
SPIClass hspi(HSPI);
#define TFT_RST 26
#define TFT_RS  25
#define TFT_CS  15  // SS
#define TFT_LED 0   // 0 if wired to +5V directly
// Use hardware SPI
TFT_22_ILI9225 tft(TFT_RST, TFT_RS, TFT_CS, TFT_LED);
```
```C
void setup() {
  hspi.begin();
  tft.begin(hspi);
}

void loop() {
  tft.setDisplay(true);
```

Bolukan, 2018-11-04:
I tested the Basic Demo with a D1 mini, Arduino 1.8.6 and TFT_22_ILI9225 1.3.8. Results ok.
Wires attached like see below and changed some code for the correct pin numbers (second block).

```C
// ILI9225 (new variant) -> ESP8266 D1 MINI
// 3V3 -> 3V3
// 5V  -> not attached to ESP8266
// GND -> GND
// SCK -> D5/IO 14 = SPI-SCK
// SDO -> D6/IO 12 = SPI-MISO
// SDI -> D7/IO 13 = SPI-MOSI
// SS  -> D8/IO 15 = SPI-SS0
// RS  -> D1/IO 5
// RST -> D2/IO 4
// BL  -> D4/IO 2 but 5V works also
```
```C
#define TFT_RST 4
#define TFT_RS  5
#define TFT_CS  15  // SS
#define TFT_SDI 13  // MOSI
#define TFT_CLK 14  // SCK
#define TFT_LED 2   // 0 if wired to +5V directly
```

RanTalbott, 2018-11-05:
Hmmm, I'm using D0 instead of D8 for TFT_CS. Maybe this library isn't overriding the default properly, but the other one is???

I'll try switching that (a bit of a hassle: I used a prototype shield to solder up a nice little pinout adapter so I could use a 1-to1 cable to connect to the display), and try updating the IDE software to see whether there's a bug there. Won't be for at least a few days, though: I have a new project starting up at work that will keep me busier than usual.

Thanks for the help.

Nkawu, 2018-11-05:
Sorry, I have also been very busy at work, will try to get caught up with this issue soon.

Bolukan, 2019-02-18:
Please comment whether the bug still exist with the latest code. If not (which I guess) this issue could be closed