Nkawu/TFT_22_ILI9225 #43: Bug in fastmode

Bolukan, 2018-12-22:
Hi again,

I discovered another bug. drawChar applies startWrite/checkSPI but if fastmode becomes enabled, the code calls _setWindow, which also calls startWrite. Hence a hangup.

Before proposing a pull request (currently I disable fastMode), I wonder how we are supposed to deal with these startWrite/checkSPI functions. 
The current idea is that only the first function opens the SPI transaction. But as also helper functions can be called directly themselves (p.e. drawText calls drawChar, drawChar calls _setWindows, etc), the code has to check whether it is a main function or a helper function. 

@Nkawu (or others): how to proceed?

Nkawu, 2019-01-06:
Yes please do. Thank you for taking the lead on this, unfortunately I don't have too much time to spend on personal projects any more.

Bolukan, 2019-02-18:
solved see #44

closed 2019-02-18 by Bolukan