Nkawu/TFT_22_ILI9225 #45: Color bitmap drawing

msx80, 2019-02-08:
Hi there, i'm trying to draw a color image on the screen with some drawBitmap variants. I noticed that it takes a 2D matrix, but it might be useful to have a version with takes a single array (of size w*h*2). For example in GIMP you can save a file in C format and it outputs a ready made 16 bit 5,6,5 array.

Should be trivial to implement.

Thanks!

Nkawu, 2019-02-18:
If you want to attempt to do this, please make the code changes and create a pull request. Thanks!

lordmundi, 2022-08-06:
was just looking for this exact feature. this would be really great to have so that you don't have to load up an entire gfx library and an SD library or something if all you need to do is define the color bitmap in memory and draw it. you could even procedurally draw the color bitmap in memory as well.

lordmundi, 2022-08-07:
just an update here... i think this DOES exist but perhaps it is just not documented? In the TFT_22_ILI9225.cpp file I see a few functions like this defined:

```
//High speed color bitmap
void TFT_22_ILI9225::drawBitmap(uint16_t x1, uint16_t y1, const uint16_t** bitmap, int16_t w, int16_t h) {

```

Nkawu, 2022-08-09:
@msx80 does the GIMP export look like this?
```c
/* GIMP RGBA C-Source image dump (Cassandra.c) */

static const struct {
  guint      width;
  guint      height;
  guint      bytes_per_pixel; /* 2:RGB16, 3:RGB, 4:RGBA */ 
  guint8     pixel_data[300 * 549 * 4 + 1];
} gimp_image = {
  300, 549, 4,
  "\377\377\377\377\377\377\377\377\377\377\377\377\377\377\377\377\377\377"
...
,};
```