Nkawu/TFT_22_ILI9225 #59: License

deltabeard, 2019-07-07:
Clarification from @4dsystems is required regarding the license of this project, as I have not found any information regarding the use of this code.

Thanks.

deltabeard, 2019-07-11:
According to correspondence with @4dsystems, the repository at https://github.com/4dsystems/screen_4D_22_library is released under a public domain license, and is free for any use. For clarity, "The Unlicense" license may be used as a definition of the terms of the original repository.

As of fd25cebe0c94422f95a22f9ffcfd62e5f281e5d8, this repository is licensed under GPLv3, which means that any software that uses this code must also be licensed under GPLv3. @Nkawu LGPL may have been more suitable?

Nkawu, 2019-07-11:
Just curious - what is your interest i this matter?

deltabeard, 2019-07-11:
I want to use this library in my project, but because there was no license before, I legally wouldn't actually be allowed to. So I would then have to have writen my own library from scratch using the ILI9225 datasheet.

Now that it's GPLv3, my project would be forced to me GPLv3 too, instead of MIT or Noncommercial, for example. But it's totally up to you what license you choose.

Just because it's open source, doesn't mean it's free software. I just wanted to make sure that I would be able to use the code or not, that's all. 🙂

closed 2020-11-09 by Nkawu