Nkawu/TFT_22_ILI9225 #61: Hardware SPI on ESP32

dl9rdz, 2019-10-07:
ESP32 supports hardware SPI on arbitrary GPIO pins.

Could you add a feature that allows using hardware SPI instead of software SPI in the code?(instead of activating software SPI for non-standard pins, the pin numbers need to be passed to _spi.begin(...) as arguments)

A quick&dirty hack solution (for ESP32 only, tested and functional) is to replace
```
// Hardware SPI
SPI_BEGIN();
```
with 
```
// Hardware SPI
if(_clk < 0) { SPI_BEGIN();  } 
else {
    SPI_OBJECT.begin(_clk, -1, _sdi, _cs);
    _clk = -1;  // force use of hardware SPI in remaining code parts
}
```

But this is not elegant, and of course should be made conditional only for ESP32

MrEliptik, 2020-10-26:
Hi @dl9rdz , I'm also using an ESP32 and I'd like to use the hardware SPI. While there's no real implementation, I'd like to use your method, but I can't figure out how. I've replace `SPI_BEGIN();` with your code, and I'm still initializing the TFT_22_ILI8225 object with the SDI and CLK pin, as I understand we specify those pins in the else statement. But using that, the display stays white, like there's no communication. 
I'm using `TFT_RST=4, TFT_RS=19, TFT_CS= 5, TFT_SDI=23,#define TFT_CLK=18` (VSPI pins). 
Object declaration `TFT_22_ILI9225 tft = TFT_22_ILI9225(TFT_RST, TFT_RS, TFT_CS, TFT_SDI, TFT_CLK, TFT_LED);`

curcius, 2025-01-17:
Any solution? I'm using the ili9225 in conjunction with the esp32 in a dsp radio with the si4732 but the HSPI causes a lot of noise in the higher bands. I'm trying to make it work with VSPI but without success, only a white screen appears. If I don't 😪😪😪