Nkawu/TFT_22_ILI9225 #72: Support STM32 Nucleo boards (STM32L412KB)

pierbout, 2020-11-09:
Has this library been tested successfully with STM32 Nucleo boards? 

It works fine with an Arduino Nano, but I don't manage to run the examples on a STM32L412KB, which has the same pins. It compiles but nothing happens after loading. even a Serial.println message doesn't show off, which suggests that it fails earlyer.

I tried to declare the pins with Arduino style (9, 10, 11...) and STM style (PA8, PA11, PB5).

Thank you for any help.

#include "SPI.h"
#include "TFT_22_ILI9225.h"

#define TFT_RST PB4     //12
#define TFT_DC  PA8     //9
//#define TFT_RS  PA8     //9
#define TFT_CS  PA11    //10  // SS
#define TFT_SDI PB5     //11  // MOSI
#define TFT_CLK PB3     //13  // SCK
#define TFT_LED 0   // 0 if wired to +5V directly

TFT_22_ILI9225 tft = TFT_22_ILI9225(TFT_RST, TFT_DC, TFT_CS, TFT_LED, TFT_BRIGHTNESS);

void setup() {
  
  Serial.begin(9600);
  Serial.println("Setup...");
  
//#if defined(ESP32)
//  hspi.begin();
//  tft.begin(hspi);
//#else
//  tft.begin();
//#endif
//
//  //Orientation paysage
//  tft.setOrientation(3);
//  tft.clear();

  Serial.println("Setup end...");
}

Nkawu, 2020-11-09:
Not as far as I know. Officially this library was only intended for Arduino boards, but a lot of changes have been contributed to make it work on other boards. If you do get it to work (or if someone has), please create a pull request with the changes.

pierbout, 2020-11-09:
Thanks a lot for the quick answer. Sad to read that.

I don't think I'll have time to work on it, but if I find a solution we'll keep in touch.