Nkawu/TFT_22_ILI9225 #77: Compatibilidad con raspberry pi pico

Luuciano27, 2021-05-19:
Hola, estoy teniendo problemas para usar esta libreria con raspberry pi pico, uso platformio en visual studio code, los errores son los siguientes:

```
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:11:30: error: unknown type name 'uint8_t'
   11 |   #define fontdatatype const uint8_t
      |                              ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:20:1: note: in expansion of macro 'fontdatatype'
   20 | fontdatatype Terminal6x8[] PROGMEM = {
      | ^~~~~~~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:20:28: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'PROGMEM'
   20 | fontdatatype Terminal6x8[] PROGMEM = {
      |                            ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:11:30: error: unknown type name 'uint8_t'
   11 |   #define fontdatatype const uint8_t
      |                              ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:128:1: note: in expansion of macro 'fontdatatype'
  128 | fontdatatype Terminal11x16[] PROGMEM = {
      | ^~~~~~~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:128:30: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'PROGMEM'
  128 | fontdatatype Terminal11x16[] PROGMEM = {
      |                              ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:11:30: error: unknown type name 'uint8_t'
   11 |   #define fontdatatype const uint8_t
      |                              ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:236:1: note: in expansion of macro 'fontdatatype'
  236 | fontdatatype Terminal12x16[] PROGMEM = {
      | ^~~~~~~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:236:30: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'PROGMEM'
  236 | fontdatatype Terminal12x16[] PROGMEM = {
      |                              ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:11:30: error: unknown type name 'uint8_t'
   11 |   #define fontdatatype const uint8_t
      |                              ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:344:1: note: in expansion of macro 'fontdatatype'
  344 | fontdatatype Trebuchet_MS16x21[] PROGMEM = {
      | ^~~~~~~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\DefaultFonts.c:344:34: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'PROGMEM'
  344 | fontdatatype Trebuchet_MS16x21[] PROGMEM = {
      |                                  ^~~~~~~
Compiling .pio\build\pico\FrameworkArduinoVariant\double_tap_usb_boot.c.o
*** [.pio\build\pico\libe47\TFT 22 ILI9225\DefaultFonts.c.o] Error 1
In file included from src\main.cpp:4:
.pio\libdeps\pico\TFT 22 ILI9225\src/TFT_22_ILI9225.h:419:18: error: 'RwReg' does not name a type
  419 |         volatile RwReg *mosiport, *clkport, *dcport, *rsport, *csport;
      |                  ^~~~~
*** [.pio\build\pico\src\main.cpp.o] Error 1
In file included from .pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp:1:
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.h:419:18: error: 'RwReg' does not name a type
  419 |         volatile RwReg *mosiport, *clkport, *dcport, *rsport, *csport;
      |                  ^~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp: In member function 'void TFT_22_ILI9225::begin()':
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp:281:5: error: 'csport' was not declared in this scope; did you mean 'ushort'?
  281 |     csport    = portOutputRegister(digitalPinToPort(_cs));
      |     ^~~~~~
      |     ushort
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp:281:17: error: 'portOutputRegister' was not declared in this scope
  281 |     csport    = portOutputRegister(digitalPinToPort(_cs));
      |                 ^~~~~~~~~~~~~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp:282:17: error: 'digitalPinToBitMask' was not declared in this scope; did you mean 'digitalPinToPinName'?
  282 |     cspinmask = digitalPinToBitMask(_cs);
      |                 ^~~~~~~~~~~~~~~~~~~
      |                 digitalPinToPinName
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp:283:5: error: 'dcport' was not declared in this scope
  283 |     dcport    = portOutputRegister(digitalPinToPort(_rs));
      |     ^~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp:294:9: error: 'clkport' was not declared in this scope
  294 |         clkport     = portOutputRegister(digitalPinToPort(_clk));
      |         ^~~~~~~
.pio\libdeps\pico\TFT 22 ILI9225\src\TFT_22_ILI9225.cpp:296:9: error: 'mosiport' was not declared in this scope
  296 |         mosiport    = portOutputRegister(digitalPinToPort(_sdi));
[... 278 lines cut ...]

  // Draw first string in big font
  String s1 = "6789";
  tft.setGFXFont(&FreeSans24pt7b); // Set current font
  tft.getGFXTextExtent(s1, x, y, &w, &h); // Get string extents
  y = h; // Set y position to string height
  tft.drawGFXText(x, y, s1, COLOR_RED); // Print string

  // Draw second string in smaller font
  tft.setGFXFont(&FreeSans12pt7b);  // Change font
  String s2 = "Hello"; // Create string object
  tft.getGFXTextExtent(s2, x, y, &w, &h); // Get string extents
  y += h + 10; // Set y position to string height plus shift down 10 pixels
  tft.drawGFXText(x, y, s2, COLOR_BLUE); // Print string
  
  // Draw third string in same font
  String s3 = "World!"; // Create string object
  y += h + 2; // Set y position to previous string height plus shift down 2 pixels
  tft.drawGFXText(x, y, s3, COLOR_GREEN); // Print string
}

// Loop
void loop() {
}
```

Nkawu, 2021-05-19:
Unfortunately I have never used the library on any hardware platform than Arduinos, most have been contributed by other people. Hopefully someone can comment and help?

_**Google translated:**
Hello, I am having problems using this library with raspberry pi pico, I use platformio in visual studio code, the errors are the following:_

Nkawu, 2022-06-21:
Great work @Dresch123 ! I'll attempt to integrate the changes into the main library.